Ultimaker/Ultimaker2Marlin #59: 15.01-RC8

melf1, 2015-01-15:
Hi,
I installed Cura 15.01-RC8 and uploaded the default firmware to an Ultimaker 2. Changing the filament doesn't work properly. The stepper motor that feeds the material goes bonkers:

Removal: https://www.youtube.com/watch?v=1_Ut64OuOzs
Insertion: https://www.youtube.com/watch?v=D8_MJxpzQIc

I switched back to the firmware that is shipped with cura 14.12.1 and the behaviour disappeared immediately.

daid, 2015-01-15:
Yes, noticed the same on a few machines in the office. This is the result of me fixing another bug, causing the speed of the motor to be higher in the insertion/removal step. (it no longer "zigs/zags" now)

melf1, 2015-01-19:
The commit e4f044e9244342cc8e51011114e414009b271e10 doesn't fix the problem on my machine. After a bit of trial and error I decreased both FILAMENT_INSERT_FAST_SPEED and FILAMENT_REVERSAL_SPEED to 130.

commit d175316d (2015-01-19): So this should fix #59
closed 2015-01-19 by daid