Ultimaker/Ultimaker2Marlin #60: I cannot get it to compile on any other PCB board other than #72

bomtrek, 2015-01-21:
I selected board #7 and  REPRAP_DISCOUNT_SMART_CONTROLLER for the LCD 
and it give me this error:

ConfigurationStore.cpp: In function 'void Config_ResetDefault()':
ConfigurationStore.cpp:288: error: 'motor_current_setting' was not declared in this scope

daid, 2015-01-21:
Duh. This firmware is strictly for the Ultimaker2. You want to look at:
https://github.com/MarlinFirmware/Marlin/

closed 2015-01-21 by daid

bomtrek, 2015-01-21:
Sorry I didn't know that. I am building UM2 I just don't have original PCB and Oled display.

bomtrek, 2015-01-21:
One more question :-) .. Which one is better for me then..

this one https://github.com/MarlinFirmware/Marlin/  
or
this one https://github.com/Ultimaker/Marlin ?

daid, 2015-01-21:
Top one. The other one is intended for the UMO.

bomtrek, 2015-01-21:
So in Cura I select I have  .. UMO UM+ or UM2 ?

I am building UM2 from scratch, but everything will be exactly the same as specs. except for the PCB and Oled.