Ultimaker/Ultimaker2Marlin #69: Possible false positive temp sensor errors

illuminarti, 2015-03-01:
A couple of users (one for me, one for the Swedish support group) have reported printers that previosuly worked fine but then immediately refuse to turn on, after upgrading to the firmware that came with Cura 15.02.  Both gave 'Error - Stopped Temp Sensor' errors. 

In both cases, the behavior continued after downgrading to firmware in Cura 14.12, but went away when they downgraded to firmware from 14.09.

At least one of these users went through the process a couple of times, and was able to replicate it.

daid, 2015-03-02:
Strange, nothing was really modified in the temperature sensing&temp error code between those versions...

I guess you don't know what the serial output says in this case? (as that can help to diagnose the problem)

illuminarti, 2015-03-03:
One of the users who had this happen is going to try and  reproduce it and  capture serial output. Anything in particular to do/look for?

daid, 2015-03-03:
Just wondering what the temperature it is when it reports the error (think it will report that with the error) because certain values could explain a lot while other can raise more questions.