Ultimaker/Ultimaker2Marlin #71: Marlin lite uses M142 for RGBW LED coloring, this M code is allready used, M420 is intended for this.

oliv3r, 2015-05-19:
Over on the reprap gcode wiki, it is states that M142 is used for Holding Pressure: http://reprap.org/wiki/G-code#M142:_Holding_Pressure

For RGB coloring, M420 is already reserved: http://reprap.org/wiki/G-code#M420:_Set_RGB_Colors_as_PWM
For example  M420 R255 E255 B255 (all caps as all gcodes), however W is missing here, which we can extend the current definition with.

The only weirdness here is that E is used rather then G, as G is normally used for regular GCodes. I'm not educated enough to claim to know if this actually holds true.

daid, 2015-05-19:
FYI: The reprap wiki is a fucking mess with the GCode page. It's full of errors and mistakes.

Nasimovy, 2020-05-04:
according to marlin wiki https://marlinfw.org/docs/gcode/M150.html
M150 should be used for RGB(W)
and M420 is used for bed leveling state https://marlinfw.org/docs/gcode/M420.html