Ultimaker/Ultimaker2Marlin #72: Marlin-lite panics when using M42

oliv3r, 2015-05-19:
When an M42 is called in a marlin-lite equipped board, the board fails to respond and the watchdog timer responds after a while.

ErikDeBruijn, 2015-05-20:
I had exactly the same thing with an UltiMainboard. Using my own code to set the same I/O pin as M42 was supposed to set, I had no crashes or watchdog timer issues.

Although the problem isn't affecting me anymore if it's becomes clear why it occurs, I'm interested to know.

ckielstra, 2017-05-15:
The error reports don't specify the exact M42 commands given. Since this command filters about 20 pins that are not allowed for changes, it is impossible to verify the reports. Also note that the M42 command outputs a PWM signal, so instead of 0/1 you have to specify 0/255 to get constant outputs.

October 2015 a few small fixes were implemented for the M42 command.

There have been no follow up reports in the last 2 years and since the error is reported in a non-reproducable way I close the ticket.

closed 2017-05-15 by ckielstra