Ultimaker/Ultimaker2Marlin #73: Ultimaker 2 firmware not fault tolerant

bburrough, 2015-05-21:
I was trying to confirm that I would get "unknown command" for M200, but...

SENT: M200 S0.0
RECEIVED: ok
SENT: M1024
RECEIVED: ok
SENT: M9999
RECEIVED: ok
SENT: M99997654321
SENT: M99997654321
SENT: M999976
SENT: M99999
SENT: M9999
SENT: G28

The firmware went out to lunch after I entered M99997654321.  It eventually stopped responding altogether (which is to say I couldn't enter any further commands).

Then, after I restarted the printer and reconnected, I entered MGOBBLEDYGOOK and nozzle moved all the way forwarded until it collided with the front of the printer!

The reason I think this is an issue that should be fixed is:

1) The printer should not allow itself to be damaged due to bad input.
2) I suspect this is part of the problem with USB printing.  That is, if any part of communication fails, the firmware can break utterly.

I understand that garbage in means garbage out, but the firmware should validate its input before executing it, otherwise, printer damaged.  :(

bburrough, 2015-05-21:
Further, if I'm right, printing from the SD card doesn't make this go away.  A flaky SD card could cause the same thing to occur.

daid, 2015-05-21:
1) While not sounding very nice, it does not really damage the machine. Only run away temperature can damage the machine, and it's protected against that.
2) USB printing, while working, is not something we officially endorse. You are free to play with it, but no guarantees. Note that normal GCode streamers use checksums which prevent transmissions errors.
3) SD card reads are done with CRC checks.
4) People are free to experiment with their machines, and free to do strange and weird stuff. Because of this, we do not lock everything down for a reason.