Ultimaker/Ultimaker2Marlin #74: Trying to verify and compile? Has this happened to you before?

FlintSable, 2015-06-12:
Arduino: 1.6.4 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Wire\utility\twi.c.o: In function `__vector_39':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\utility/twi.c:364: multiple definition of `__vector_39'
UltiLCD2_low_lib.cpp.o:C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp/UltiLCD2_low_lib.cpp:184: first defined here
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions
collect2.exe: error: ld returned 1 exit status
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

FlintSable, 2015-06-12:
Arduino: 1.6.4 (Windows 7), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Build options changed, rebuilding all

Using library Wire in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 

Using library LiquidCrystal in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 

Using library SPI in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\cardreader.cpp -o C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\cardreader.cpp.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\ConfigurationStore.cpp -o C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\ConfigurationStore.cpp.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\electronics_test.cpp -o C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\electronics_test.cpp.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\lifetime_stats.cpp -o C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\lifetime_stats.cpp.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\LiquidCrystalRus.cpp -o C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\LiquidCrystalRus.cpp.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\Marlin.cpp -o C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\Marlin.cpp.o 

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++ -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10604 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\mega -IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire -IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src -IC:\Program F [... cut ...]
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\utility/twi.c:364: multiple definition of `__vector_39'
C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp\UltiLCD2_low_lib.cpp.o:C:\Users\NICHOL~1\AppData\Local\Temp\build6992659436616663863.tmp/UltiLCD2_low_lib.cpp:184: first defined here
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions
collect2.exe: error: ld returned 1 exit status
Error compiling.

FlintSable, 2015-06-12:
I am sorry about that long one. Do you have an idea of why I cannot compile.

Dim3nsioneer, 2015-06-13:
Move the wire library temporarily to a different location where the Arduino compiler cannot find it. Then recompile it. I think to remember that there was a second library with the same problem.

daid, 2015-06-13:
See the note at:
https://github.com/Ultimaker/Ultimaker2Marlin/blob/master/Marlin/Configuration.h#L419

closed 2015-06-13 by daid

FlintSable, 2015-06-15:
I spent a few days looking for this function and didn't exactly find what I needed. Thank you for your reply. 

when looking for this function 
SIGNAL(TWI_vect)" function from "libraries/Wire/utility/twi.c

I looked here C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\twi.c
and found this function. 

I deleted this function and was not able to compile still. Is there anything else I should be aware of. 
Is it ok that I am working on WINDOWS 7. I can try working with the platform you suggest.

ISR(TWI_vect)
{
  switch(TW_STATUS){
    // All Master
    case TW_START:     // sent start condition
    case TW_REP_START: // sent repeated start condition
      // copy device address and r/w bit to output register and ack
      TWDR = twi_slarw;
      twi_reply(1);
      break;

    // Master Transmitter
    case TW_MT_SLA_ACK:  // slave receiver acked address
    case TW_MT_DATA_ACK: // slave receiver acked data
      // if there is data to send, send it, otherwise stop 
      if(twi_masterBufferIndex < twi_masterBufferLength){
        // copy data to output register and ack
        TWDR = twi_masterBuffer[twi_masterBufferIndex++];
        twi_reply(1);
      }else{
	if (twi_sendStop)
          twi_stop();
	else {
	  twi_inRepStart = true;	// we're gonna send the START
	  // don't enable the interrupt. We'll generate the start, but we 
	  // avoid handling the interrupt until we're in the next transaction,
	  // at the point where we would normally issue the start.
	  TWCR = _BV(TWINT) | _BV(TWSTA)| _BV(TWEN) ;
	  twi_state = TWI_READY;
	}
      }
      break;
    case TW_MT_SLA_NACK:  // address sent, nack received
      twi_error = TW_MT_SLA_NACK;
      twi_stop();
      break;
    case TW_MT_DATA_NACK: // data sent, nack received
      twi_error = TW_MT_DATA_NACK;
      twi_stop();
      break;
    case TW_MT_ARB_LOST: // lost bus arbitration
      twi_error = TW_MT_ARB_LOST;
      twi_releaseBus();
      break;

    // Master Receiver
    case TW_MR_DATA_ACK: // data received, ack sent
      // put byte into buffer
      twi_masterBuffer[twi_masterBufferIndex++] = TWDR;
    case TW_MR_SLA_ACK:  // address sent, ack received
      // ack if more bytes are expected, otherwise nack
      if(twi_masterBufferIndex < twi_masterBufferLength){
        twi_reply(1);
      }else{
        twi_reply(0);
      }
      break;
    case TW_MR_DATA_NACK: // data received, nack sent
      // put final byte into buffer
[... 80 lines cut ...]
      TWDR = twi_txBuffer[twi_txBufferIndex++];
      // if there is more to send, ack, otherwise nack
      if(twi_txBufferIndex < twi_txBufferLength){
        twi_reply(1);
      }else{
        twi_reply(0);
      }
      break;
    case TW_ST_DATA_NACK: // received nack, we are done 
    case TW_ST_LAST_DATA: // received ack, but we are done already!
      // ack future responses
      twi_reply(1);
      // leave slave receiver state
      twi_state = TWI_READY;
      break;

    // All
    case TW_NO_INFO:   // no state information
      break;
    case TW_BUS_ERROR: // bus error, illegal stop/start
      twi_error = TW_BUS_ERROR;
      twi_stop();
      break;
  }
}

FlintSable, 2015-06-16:
[image]

FlintSable, 2015-06-16:
Has the location of that function changed?

FlintSable, 2015-06-16:
[image]

FlintSable, 2015-06-16:
when I delete that function I get this error but it is done. 

Is this safe?

FlintSable, 2015-06-16:
Thank you again for taking the time to help.

rajanjamu, 2015-07-08:
I am also getting this exact same error? Any workarounds for this?