Ultimaker/Ultimaker2Marlin #76: How to display Chinese language to the oled menu? Tks.

nxla, 2015-07-05:


oliv3r, 2015-07-05:
You write code for it. Mind you, this is not an easy task, there is no font support and none of the strings account for it. Everything you need is in the UltiLCD2 files. Looking forward to your pull-request or a link to your repository to see how you solved it.

Good luck!
Olliver

closed 2015-07-05 by oliv3r

nxla, 2015-07-05:
Thank you. Yes,we have Try doing it. We use 12*12 fronts, but anyhow, it's only display 8 pixels in height.

xccount, 2015-07-06:
Can I mail you code to check the problem? Thank you for your help.

Regards. 

W

daid, 2015-07-06:
How much will you pay?

xccount, 2015-07-06:
How much you think its reasonable？ we will consider your quotation.
 
 
------------------ Original ------------------
From: "daid"; 
Date: 2015年7月6日(星期一) 下午4:56
To: "Ultimaker/Ultimaker2Marlin"; 
Cc: "xccount"; 
Subject: Re: [Ultimaker2Marlin] How to display Chinese language to the oledmenu? Tks. (#76)

 

How much will you pay?