Ultimaker/Ultimaker2Marlin #77: Low memory available, stability problems may occur.

rajanjamu, 2015-07-09:
Arduino version: 1.6.5
OS: OS X Yosemite - Version 10.10.1

Removed function
ISR(TWI_vect) 
from 
\hardware\arduino\avr\libraries\Wire\twi.c

Arduino 1.6.5 output
----------------------------------------------------------------------------------------------------------------
Sketch uses 116,414 bytes (45%) of program storage space. Maximum is 253,952 bytes.
Global variables use 6,314 bytes (77%) of dynamic memory, leaving 1,878 bytes for local variables. Maximum is 8,192 bytes.
Low memory available, stability problems may occur.
----------------------------------------------------------------------------------------------------------------

norpchen, 2015-07-09:
This is a generic warning from the Arduino compiler that has no understanding how much dynamic memory may or may not be allocated by your sketch at runtime.  Marlin firmware is frugal with dynamic allocations and local variables, and Ultimaker's branch continues this practice.  It's pretty conservative -- 1878 bytes is plenty of free ram.

FWIW, my fork of the UM2 firmware is much more aggressive -- it uses almost 7500 bytes.  I find memory problems don't really show up with this firmware until you get under 500 bytes free

daid, 2015-07-10:
Yes, we're quite fine as @norpchen points out. As the only dynamic memory we use is for the stack. (We're not using heap memory)

Also, the official builds use a slightly different build script, saving off 100 bytes:
```
Program:  125922 bytes (48.0% Full)
(.text + .data + .bootloader)
Data:       6204 bytes (75.7% Full)
(.data + .bss + .noinit)
```

closed 2015-07-10 by daid