Ultimaker/Ultimaker2Marlin #98: Reset when entering Print menu

restinpieces, 2016-02-14:
I updated from somewhere around summer 2014 to the current version and discovered that Ultimaker2Marlin resets every time I try to enter the print menu.
I tracked the bug back to the commit on the 20. January "Lower the change material temperature depending on the material"
https://github.com/Ultimaker/Ultimaker2Marlin/commit/c2ab36848c5d2e4097a8add289265948d14c15cc
The commit before is fine.
The "Lower the change material temperature depending on the material" seems to introduce this bug. In all later versions I'm unable to enter the print menu.

I first thought it has something to do with reading the SD card, but importing and exporting material settings to  SD card still works. So no clue what the problem really is.

restinpieces, 2016-05-08:
Am I the only one with this problem? I still can't update.

daid, 2016-05-08:
As far as I know, this commit fixes it:
https://github.com/Ultimaker/Ultimaker2Marlin/commit/7408e32de2d4348a077c6b274af3635f022e1e6a

If you build with the Makefile. Cura 15.04.6 contains this fix.

ckielstra, 2018-02-19:
Closed the issue as a fix was implemented and no repeated reports of this bug were reported since.

closed 2018-02-19 by ckielstra