Ultimaker/Ultimaker2Marlin #102: Extended lcd_font_7x5 does not work

karabas, 2016-04-24:
I want 0x80-0xff chars to dispay for future cyrillization. F.e. I added a string for 0x80 char to the end of lcd_font_7x5[]. But display shows garbage.  Chars 0x80 and higher are addressed out of lcd_font string.

karabas, 2016-05-09:
The problem is that char is signed int. So I recommend to replace it to unsigned char:
 -   for( char c = pgm_read_byte(pstr); c; c = pgm_read_byte(++pstr))
 +  for(unsigned char c = pgm_read_byte(pstr); c; c = pgm_read_byte(++pstr))

karabas, 2016-06-01:
Ok, no code changing required. Problem is solved adding Arduino IDE gcc flag -funsigned-char.

arduino\hardware\arduino\avr\platform.txt

compiler.cpp.flags=-c -g -Os {compiler.warning_flags} -std=gnu++11 -funsigned-char -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD

karabas, 2016-06-01:
You may note it in docs if you want.

closed 2016-06-01 by karabas

Ghostkeeper, 2016-06-02:
Perhaps something for the build script?