Ultimaker/Ultimaker2Marlin #104: M42 will ignore FAN_PIN

derbroti, 2016-04-25:
Hi,

The fan control with M42 will not work because FAN_PIN is a "sensitive" one (sorry but the naming is not intuitive...) see:
https://github.com/Ultimaker/Ultimaker2Marlin/blob/master/Marlin/pins.h#L2154
and the loop changes pin_number to -1, which will prevent the == FAN_PIN check.

So either remove it from that list or move the fanspeed code:
https://github.com/Ultimaker/Ultimaker2Marlin/blob/master/Marlin/Marlin_main.cpp#L1327

in front of the sensitive pin checking loop here: 
https://github.com/Ultimaker/Ultimaker2Marlin/blob/master/Marlin/Marlin_main.cpp#L1319

I would provide a patch if one can tell me if FAN_PIN should be in the sensitive list or not.

daid, 2016-04-26:
It should be, it's controlled by M106/M107.

closed 2016-04-26 by daid

derbroti, 2016-04-26:
But then this part is useless:
https://github.com/Ultimaker/Ultimaker2Marlin/blob/master/Marlin/Marlin_main.cpp#L1327

#if defined(FAN_PIN) && FAN_PIN > -1
        if (pin_number == FAN_PIN)
          fanSpeed = pin_status;
      #endif

daid, 2016-04-26:
Indeed, that piece of code is useless.

derbroti, 2016-04-26:
Alright, is there any effort on removing unused / dead code? 
In another issue you said this fork is specifically only for the UM2 - but still there are the configuration directives for other printers.

Would it be welcomed to clean up the code in the sense of removing all "dead" parts and configs that are related to other printers?

daid, 2016-04-26:
Nope, there is no effort made in removing dead code. Simple from a risk perspective we're keeping this code as much as it is.