Ultimaker/Ultimaker2Marlin #105: Changes Require from Main Marlin Branch

CCS86, 2016-05-03:
Hi guys,

Is there a comprehensive list somewhere of the changes needed to adapt the main branch of Marlin for use on an Ultimaker Original?

I want to test the newer RC builds, and think they are relatively close, but want to make sure I hit all the important things.

Thanks!

daid, 2016-05-04:
Well, you ask this question on the Ultimaker2 version. The UMO firmware is here:
https://github.com/Ultimaker/Marlin
With the UMO+ here:
https://github.com/Ultimaker/Marlin/tree/Marlin_UM_Original_Plus

The UMO firmware wasn't that different from the mainline Marlin at the time of the fork. The UM2 firmware is quite a different beast.

I merged the pin settings to the mainline a long time ago, so in theory it should still work with the right motherboard config.

CCS86, 2016-05-04:
Crap, my mistake. I did indeed mean UMO. 

You say it wasn't that different, which I suspected. Is there an actual list of changes? It may very well be accessible via Github, but I'm not very familiar.

daid, 2016-05-04:
The changes are trivial, biggest one is the added CRC check on SD card reading. (Which might have been done in main Marlin as well these days)

Note that the Ultimaker version is very old compared to the main, but it's also stable. As the Marlin developers where focusing more on features then stability, I stopped following the development.

CCS86, 2016-05-04:
Thanks for the help Daid.

I've got a document compiling all the customizations I have made to my Marlin build. To test the new RC6 build, I can look for all those settings in the new version and apply. Besides that, just look for stuff related to the 2560 board and overall machine config? Any land mines you can think of?

daid, 2016-05-04:
Just make sure the watchdog is enabled. And know that RCs from Marlin have been extremely unstable in the past. More alpha quality then an RC.

If you have an UMO (not an UMO+) then make sure all errors related to the temperature sensor failing trigger correctly. Else you might have a fire risk on your hand.

karabas, 2016-05-10:
> Note that the Ultimaker version is very old compared to the main, but it's also stable. As the Marlin >developers where focusing more on features then stability, I stopped following the development.

Daid, what version of marlin do you base on?

daid, 2016-05-10:
The UM2+ is based on the UM2, the UM2 is based on the UMO. The UMO is based on a version of Marlin from about 3 years ago.

closed 2016-12-15 by ckielstra