Ultimaker/Ultimaker2Marlin #107: Feature Request - Pin code protected maintenance menu

fluetke, 2016-05-11:
Hey, 

as technical manager of a fablab, I would love to have a possibility to lock the maintenance menu on the ultimaker 2 and umo with a pin or passcode, to prevent visitors from messing with our machines.

Dim3nsioneer, 2016-05-16:
What about generating a gcode file which resets all parameters to what they should be?

fluetke, 2016-05-19:
Well, we'd have to go through every machine to load up the gcode every day, to fix possible changes the users have made, we unfortunately don't have the manpower for that. 
Also, we'd have to check the materials inserted, for abs/pla,, dust, debris etc. locking the automated filament change assistant for example, to keep users from unauthorized material changes, would already be a huge help.