Ultimaker/Ultimaker2Marlin #108: New Ultimaker 2+ firmware through Cura 15.4.5

rebekahharper, 2016-06-08:
Hiya,

I am new to the coding side of life in 3D printers.

I need to as if the latest firmware for the Ultimaker 2+ contains a new bed temperature limit of 100 degrees. for ABS and overall. 

I used to be able to use 110 degrees (normally I use 115 but couldn't on new 2+)

any help would be great. 

thank you 

Becky

ckielstra, 2016-06-08:
For safety regulations we (Ultimaker) were advised to limit the bed temperature to 100C. More users are requesting a higher temperature so the next firmware release is planned to have a 110C limit and a few minor changes to improve safety in this situation.

The 100C limitation is a new feature, so a temporary workaround would be to use a slightly older version.

rebekahharper, 2016-06-08:
Hiya,

thank you for your response. 

then that makes the new Ultimaker 2+ firmware unsuitable for ABS. I use Buildtak printbed cover to help with the adhesion. however 100 degrees wasn't enough and the part peeled off. 

110 is the recommended

I hope it is resolved. 

in the mean time, I have managed to restored the last version. and I am back were I need to be. 

Many thanks 

Rebekah

closed 2016-06-08 by rebekahharper