Ultimaker/Ultimaker2Marlin #112: Bed leveling issue

Sarchopilus, 2016-08-13:
When I go to run the initial setup the print head moves to the front center of the bed for the first adjustment instead of moving to the back center of the bed close to the adjustment screw.

ckielstra, 2016-08-16:
Which machine model and which software version did you notice this behaviour on?

We have noticed this wrong behaviour on an UM2(+ ?) Extended using firmware from the latest Cura release (v2.1.2). For unknown reasons something went wrong in the build process, we are looking into this. The hex-files from the 15.04 Cura version are ok.

Ghostkeeper, 2016-08-16:
daid came to me about that, ckielstra. We've copied Cura 15.04's firmware builds to Cura 2.1 (and master) last Thursday. The latest builds and release of Cura 2.1.3 should have correct ones.

We don't know why the firmware build had this weird behaviour. I made it with the same script as I always did. Perhaps it was just built at an inopportune moment while some fix was going on on the repository.

daid, 2016-08-17:
@Sarchopilus which version of Cura did you use to install the firmware? It would be good to know which Cura version has a bad firmware.
And which printer are you using? UM2Extended+?

RuralGeekNZ, 2016-09-08:
I found this issue with the Cura 2.1.2 firmware on a UM2+
Have not tried 2.1.3 yet.

daid, 2016-09-08:
This problem is known for the 2.1.2, the 2.1.3 does not have this problem, but does have the problem of material change during pause of a print not working correctly.

(Technical detail, we've seem to run into some kind of limitation of the CPU used, causing random bugs. We haven't had the time to track down the source yet due to us working hard towards the thing we are not allowed to tell yet)

TinkerGnome, 2016-09-10:
Y_MAX_LENGTH is not longer an integer - this leads to "undefined behavior" if it is used as a variable argument for printf.
In this specific case it enqueues
"G1 F6000 Z35 X115 **Y0**"
instead of
"G1 F6000 Z35 X115 **Y214.5**"

[Pull request 115](https://github.com/Ultimaker/Ultimaker2Marlin/pull/115)

> but does have the problem of material change during pause of a print not working correctly.

The command buffer is not empty, if the material change wizard is used during a paused print. That's why it never exits the preheat menu function

[Pull request 114](https://github.com/Ultimaker/Ultimaker2Marlin/pull/114)

ckielstra, 2017-06-16:
Closed issue since the original bed leveling problem was fixed with Cura 2.1.3. The other two referenced problems were fixed with pull request #114 and #115 on 13 September 2016 which was a long time ago.

closed 2017-06-16 by ckielstra