Ultimaker/Ultimaker2Marlin #113: Strange problem with stored material profiles after updating a PLUS upgraded UM2.

Spirou42, 2016-08-31:
After updating the firmware for my Plus upgraded UM2 (from Cura 2.1.3) I can't permanently store material profiles any more.
Loading a MATERIAL.TXT from SD-Card partially works as I can load the file and the profiles are more or less correctly loaded. Unfortunately the new per material retraction settings are not loaded correctly i.e. retraction length is read as 255 mm and retraction speed with 50mm/s for almost all materials after the defaults. I can change the setting to reasonable values but not or all retraction values and all custom material settings get dumped if i shut down the printer.

I have tried a bunch of stuff to work around the issue:
1. Reducing the number of materials in the MATERIAL.TXT  - does not work. all custom material setting get lost after shutdown. 
2 sticking to the default materials an customising them also does not work. I can store a changed preset but after shutdown the changes are lost.

So I can't create any permanent custom material settings any more.

Spirou42, 2016-08-31:
I have refreshed the firmware by downgrading to the version of Cura 2.1.2 (version 2_1) and afterwards upgrading again with Cura 2.1.3 (Version 15.04.7-RC5) which intact behaved a bit strange after upgrading. The Printer did not restart after updating. I had to turn it off and on again.

Tatsi, 2017-02-16:
I've had the exact same problem for three months. Any progress on this?

ckielstra, 2017-02-17:
The problem has been fixed and will (most likely) be included with the final Cura v2.4 release.

ckielstra, 2017-02-21:
Cura v2.4 was released today and I'm sorry to say this bugfix didn't make it. They try to release Cura more often so it shouldn't take too long. We'll keep you updated.

Tatsi, 2017-02-22:
Thanks for letting us know @ckielstra. That's unfortunate, let's hope the next Cura version gets released sooner.

daid, 2017-02-23:
https://github.com/Ultimaker/UM2.1-Firmware/commit/1cf043af597bbc3de9ef4c9cb08670ceb835b8cb

I think this is the fix for it.

Ghostkeeper, 2017-03-10:
I just updated the firmware build for Cura again to prepare for the 2.5 beta release. See https://github.com/Ultimaker/cura-binary-data/commit/6e2821dd5f52fef82e5f74dbece934a71a9c194e

There's a file that tracks the commit hashes for each of the firmware builds. It indicates that the firmware for the UM2+ just went from c5fffff0331159c640da3da9e534f77706c5df35 to 1cf043af597bbc3de9ef4c9cb08670ceb835b8cb. So the commit that daid is talking about should be included then.

The 2.5 beta is being sent to the system testing team which tests integration with other Ultimaker products. That'll take a week or two. In the meanwhile there is also an unofficial PPA held by Thopiekar, that'll make new builds of Cura with the fix included: https://launchpad.net/~thopiekar/+archive/ubuntu/cura-master

Spirou42, 2018-02-21:
Unfortunately i totally forgot about this issue after i got my UM3 and i didn't own my UM2+ any more.
But i have access to my old printer. I'll check as soon as i can if this issue can be closed.