Ultimaker/Ultimaker2Marlin #117: Retraction after material change during print?

peteruithoven, 2016-10-06:
I'm trying to print english liquorice (engelse drop), so a print with layers of different colors (and possibly material types). To pause the print I'm using the  Post Processing Plugin in Cura 2.3.0. 
I'm using the latest firmware from this repository (1dfce5af3e9ab960078c1e6664b1e01a31609946) which I compiled myself, which already includes this fix: https://github.com/Ultimaker/Ultimaker2Marlin/pull/114. 
I just noticed however that after the whole material change flow (so after you confirm the new material is flowing through the nozzle) it for some reason retracts the material a bit. I felt it retract at least 1cm filament from the feeder. This means the printer can't successfully continue printing, because it takes a while for the filament to start flowing again. 

The Retraction setting in the Pause at height plugin was set to 0.

TinkerGnome, 2016-10-10:
The retract is intended. It's the same length as the end-of-print retraction (20mm by default). You can compensate this with a proper value of "Retraction amount" (inside the post processing script).
The pause function at the printer takes care of it automatically.

For more information see:

https://ultimaker.com/en/community/22204-um2-pause-at-height-and-then-change-filament-from-screen#reply-154003

peteruithoven, 2016-10-10:
But if I understand the [PauseAtHeight script](https://github.com/nallath/PostProcessingPlugin/blob/master/scripts/PauseAtHeight.py) correctly it does 4 retraction related extrude moves, which cancel each other out. Wouldn't it need a additional extrude to compensate for that intended retract?

TinkerGnome, 2016-10-11:
The crucial point is: use the same retraction length for the post processing script and for the material change wizard - and everything works like intended.

peteruithoven, 2016-10-12:
@TinkerGnome I really appreciate the help, but now you're simply pointing out the same thing, without explaining it further. 

I opened an issue to improve the PauseAtHeight Retraction setting: https://github.com/nallath/PostProcessingPlugin/issues/23

peteruithoven, 2016-10-13:
Maybe the problem isn't that it retracts after the material change, this is actually a good safety measure making sure the material can always easily be removed and the nozzle isn't clogged. Maybe it should automatically retract again when the user presses the button to continue.