Ultimaker/Ultimaker2Marlin #119: Activating serial connectors on the board

thopiekar, 2016-12-04:
Like noticed on my visit the serial pins are not used at the moment.
Would be great to have an option somewhere to make them work like the usual USB connection, so the pins are at least not dead.

oliv3r, 2016-12-05:
To give the choice, to use USB or serial is just one configuration parameter in Configuration.h To listen to both set of incoming ports is harder, since if you use both ports, 'who is right'?

E.g. user A prints via USB, and user B tries to print via the serial port. You now have 2 incoming GCODES that will run intermingled. To handle that properly, a lot of effort is required.

thopiekar, 2016-12-07:
Well, how often would it really happen? I don't see any reason why people do such a madness and overload their printer with GCode from both sides. If people are doing crap they should get an unexpected (or whatever they were planning to do) result.
Someone who is interested in getting access via external/3rd party modules expects that there might be side-effects. She/he is obviously trying to get access this way for the first time. Additionally an USB connection is not essentially needed, eg. to power the board. At least after such an upgrade people would only need the USB port for further firmware upgrades, I guess.

Yes, it is currently possible to access the RX(1)/TX(1) pins of the active connection by soldering some test points, but that is - in my point of view - not modding-friendly at all. If this additional serial connection is really meant to be dead, why are we even wasting solder plus pins on our board and finally give these pins a name?
Don't get me wrong.. having these pins on the board is better than nothing. We often love to see people in our forums modding our printers, but I'm sure there are already some projects out there, which have been stopped, because this connectors are dead. If it is not a big deal and you really worry about people mixing up both serial I/Os, we could add a very basic switch into our printers settings to be able to choose between both serial I/Os.