Ultimaker/Ultimaker2Marlin #120: Print of area Commits on Dec 2, 2016

qwerty8224, 2016-12-09:
It happened after the addition. Commits on Dec 2, 2016
Tinkergnome Firmware has errors they should not be copied here.

daid, 2016-12-09:
?

qwerty8224, 2016-12-09:
Got error 
Error Tried printing out of printing area.
Previously, this was not

ckielstra, 2016-12-09:
I'm running the firmware here without problems.
Can you give some more details on the error you have? What did you do? What happened, and why is this different from what you expected?

qwerty8224, 2016-12-09:
After flashing the printer began to behave strangely.
At the end of the print head is not removed, and the engine began to remove the thread ....

ckielstra, 2016-12-09:
I'm not sure I understand your problem. The Dec 2 commit changes the display code and can not create your problems. Are you sure to compile the code for the correct printer? 
This is the code for the UM2. Code for the UM2+ and other Ultimaker printers is in other repositories.

I have to go home now so will be slower in responding.

qwerty8224, 2016-12-09:
Printer clone Ultimaker
Electronics Ultimaker 2.1.4

ckielstra, 2016-12-12:
Be more specific. It is a clone of which Ultimaker? The Original, Original+, the UM2 or UM2+ ?
Perhaps you have a link to the site where you bought your printer?
And then, a clone can make a lot of changes that we don't know about. 
I recommend you to contact the company you bought your printer from. Perhaps they can give you better firmware.

qwerty8224, 2016-12-12:
Printer gathered himself ....
Firmware Marlin 4mesyatsa worked without problems.
But then there was an error (((Maybe just a coincidence

ckielstra, 2016-12-12:
I don't know how to help you since many things can be different in your machine.
Did the error happen only once?

qwerty8224, 2016-12-12:
Yes once. I think that this is just a coincidence ....

closed 2016-12-12 by qwerty8224