Ultimaker/Ultimaker2Marlin #123: Head fan not controlled (e.g. not even started) during auto PID tuning

derbroti, 2017-01-29:
During the autotune the head fan is not controlled - the only place the fan is controlled is in manage_heater() which is not called during the for(;;) loop of the auto tune.

See: [temperature.cpp:554](https://github.com/Ultimaker/Ultimaker2Marlin/blob/master/Marlin/temperature.cpp#L554) and  [temperature.cpp:208](https://github.com/Ultimaker/Ultimaker2Marlin/blob/master/Marlin/temperature.cpp#L208) 

This is a problem as well for the UM2.1 repo.

Current workaround is to preheat the head before starting the PID auto tune

daid, 2017-01-30:
Word of warning, the PID autotune doesn't produce proper results for the UM2 line of printers.

derbroti, 2017-01-30:
line as in including the 2+?

daid, 2017-01-30:
Yes, UM2, UM2go, UM2extended, UM2+, UM2+extended

derbroti, 2017-01-30:
Can you elaborate on where the issue is, that it does not produce good values? I have a friend who works in the area of control systems that I wanted to asked to look at the auto tune.

daid, 2017-01-30:
Repeated runs are all over the chart in values, and the resulting values usually do not work very well and cause a lot of overshoot.