Ultimaker/Ultimaker2Marlin #130: Problem with insert material "any sounds"

Printo1987, 2017-12-11:
TOday i install this verion to the ultimaker and have any problem with insert and move out material if i wanna change material software move out material around fast 10cm and after i hear any SOUNDS and nothing else.. ON the original SOftware and Tinder Gnome i dont have this Problem..

Anybody can help me? THank you

ckielstra, 2017-12-12:
Your problem description creates many questions.
First of all, the official Ultimaker releases are part of the Cura software and can be installed from there. The software in this repository is for people who want to make special changes and you need to know what you are doing. Support given here will be minimal since you are not using the official released software.

1) Why do you want to install the software from this repository?
2) When you say you "installed this version", then which branch do you mean? I guess the master branch, but just want to make sure.
3) "any sounds" is very vague. Can you be more specific? Is it a sound from the motors like grinding, a long move, or short moves? Or is a sound from the speaker? Please give as many details as possible.

Printo1987, 2017-12-12:
Thanks you for replay :)

I never been problem with Marlin ... and I tell you everything:)

In original software by Cura I dont have this problem but I need Install This version by Marlin Arduino becuase this version is better yes? Smooth Printing and etc yes?  Original Version firmware is this same??
and If i  install this Firmware by Arduino is ok .. leveling temperature is ok everything ok but problem is only with change material or insert material I put inside Material click insert material after click again "OK" and material faster inside to hotend but only travel 5cm and after motors extruder stop and I here Sound SPekaer around 3sec and after again material inside do hotend but very very slowely. :(

Thank you for Helping.

ckielstra, 2017-12-13:
The software here on the master branch is the newest software but is not always better. This software is not fully tested yet. When tests are finished we make a new release.

Latest tested software is always included in the Cura releases. Best is you get a recent Cura version and then upgrade your printer with the firmware in that Cura version.

The last UM2  firmware update was included with Cura 2.4, February 2017. Only small bugs were fixed, no big quality improvements. See release notes at:
https://ultimaker.com/en/products/cura-software/release-notes

Printo1987, 2017-12-13:
yes I have upgrade fedders  white colour :) steps is 369...   on the firmware original by cura i don't have problem but  I see this Marlin new Version  for Ultimaker will be better because is Smoothie print.

ckielstra, 2017-12-19:
If the original Cura software is working, then please use that version. It is the latest and tested software. The software here has only a few small fixes, not improvements to the printing quality.

closed 2017-12-19 by ckielstra

Ghostkeeper, 2018-01-05:
Actually, this software hasn't changed since May 2017. The build I've made for Cura is more modern than that, so the version in Cura should be the same as when you run it yourself.