Ultimaker/Ultimaker2Marlin #131: M601 does not pause print

Chris-j-hill, 2018-02-05:
M601 is meant to pause the print from gcode and wait for the user to manually resume. When this gcode is run the nozzle should be parked and a "while(card.pause)" loop is begun. However, the value of card.pause is never set to true in M601. Result being the nozzle goes to the specified park position then returns to the print job immediately. Simple fix is to explicitly set the value true just before the while loop.

ckielstra, 2018-09-07:
Not a bug but a feature request since the M601 command is an internal command and not listed in the supported G codes.

So, the request is to make M601 a supported command...

M601 is like an improved version of the M0 command where the build plate will move down a bit and the head moves to a parking location. The PauseAtHeight plugin in Cura uses the M0 command and a few extra gcodes to do the same (and more) as the M601.

We can make the M601 externally available, but just wondering what would be your use case?

Chris-j-hill, 2018-09-07:
For context, I was using this as part of a filament run out sensor addon I was working on at the time. My system worked simply by polling a pin connected to a limit switch and then when a run out was detected run the M601 command. So all of this was kept within the firmware. 

I didnt use the M600 command because of the beeping, the printer was situated in a college with offices and lecture rooms adjacent to where the printers are located so allowing the printer to beep continuously if unattended was not an acceptable solution. M601 which just pauses and parks the nozzle seemed better, however it then didn't behave as I would have expected. 

So in my case I don't need it to be a supported command. I changed M601 at my end to explicitly set the printer to be in the paused state and it hasn't broken anything that I'm aware of. Is there something that depends on this not being set within the function? If not, surely this would just lead to unnecessary code bloat if it needs to be set in a bunch of different places whenever M601 is being used. 

Im not familiar with the PauseAtHeight plugin so I don't know what codes it's generating. I use simplify 3d as my primary slicer, and while I have not found a case where I need a pause at height feature, off the top of my head I'm not sure of a simple script I could write to achieve this without using M600, because of noise concerns, or M601, because its an internal command. In which case, maybe an alternative solution would be to support additional behavior for M600? That would be useful in my case but seems like more work than its worth, and would be to solve a pretty niche problem...