Ultimaker/Ultimaker2Marlin #139: MESH bed leveling

TomyVertigo, 2020-11-09:
Commands for mesh bed leveling not included in the configuration file. Can you please add them? If I try to copy them from the original Merlin file I get ''missing binary operator before token "(" ERROR.

Thank you

Ghostkeeper, 2020-11-17:
The Ultimaker 2 doesn't do mesh bed levelling, right? It wasn't implemented yet back then. What is it you are trying to accomplish?

TomyVertigo, 2020-11-17:
Hello, 

I try to compensate for an unstraight bed.

Ghostkeeper, 2020-11-19:
Ultimaker itself has no plans to release new versions of the Ultimaker 2 firmware unless perhaps there is a serious security issue with it. There is no development team assigned to the Ultimaker 2 at the moment. I'll leave this open for an open source community to implement if they like, but I don't think we can expect any resources to be devoted to this feature for the Ultimaker 2 any more.
The Ultimaker 2+ Connect has a different code base and is still being actively developed.